sujith90/Pi #16: Add logic for break reminder

BrianPearTree, 2016-01-14:
One of our technical requirements is to have a configurable break reminder. <br>
Add the logic to the code to require a break - after a the employee is in cubical for a certain amount of time?
This is part of our stretch requirement to improve safety.

btbam91, 2016-01-15:
@BrianPearTree Issue for the ETS reminder if you leave your cube within X minutes of a configurable time?

BrianPearTree, 2016-01-16:
If I understand your question correctly, if you have x set to 45 minutes and you leave your cube 44 minutes before ets time, an alarm will sound to remind you to fill out your time card. 

I suppose you could use a visual reminder at 45 mins whether you are in the cube or not. I'm not sure how annoying that would be, but it could be turned off.

BrianPearTree, 2016-01-16:
@btbam91 btw

btbam91, 2016-01-16:
@BrianPearTree sorry, I should really speak in complete sentences. My intent was to ask if you plan on creating an "issue" for that feature.

btbam91, 2016-01-25:
I will get this done ASAP. I do not expect it to be too difficult.

btbam91, 2016-02-03:
80% done with this feature. Sorry it's taking so long... I barely started working on it today. I drank a red bull to make sure I finish the majority of it tonight. :)

btbam91, 2016-02-04:
I am closing this issue. As we discussed, the logic is currently tied to the Break Screen in the UI. Need to integrate with Sujith to tie the logic to OMRON data.

[image]

[image]

Demo: [http://www.screencast.com/t/tyvtJomV1]

closed 2016-02-04 by btbam91