sujith90/Pi #17: Consult BDE, TLE, ATF or TF on what you are doing

BrianPearTree, 2016-01-14:
If you know someone of one of these roles outside of Manuel, consult them about a problem you are working on.

Circuit Analysis tools at Boeing - does our circuit diagram look good? Do you see any problems with it?
What sort of things should I bring to the CDR? 
Would you mind taking a look at our code and recommending improvements?
etc. Each expert we consult is worth 200 pts.

BrianPearTree, 2016-01-14:
Bill Wood is someone we know, do some searches in the corporate directory to see if there are others.

btbam91, 2016-01-28:
I talked to Mark Boyd this morning and he is willing to do a code review with me. Is there any documentation needed to have proof that the consultation actually happened?

btbam91, 2016-01-28:
Adding on to my last comment, I think Brian/Sujith should also be there since you guys worked on the OMRON and motor code.

BrianPearTree, 2016-01-29:
Okay! That sounds fine. Thanks for setting it up!

BrianPearTree, 2016-01-30:
As for proof. You might ask about a template he normally uses for code reviews. We could also search for our own form. We can add this to our documentation as proof that we consulted Mark.

BrianPearTree, 2016-02-22:
We consulted Manuel throughout, Full marks!

closed 2016-02-22 by BrianPearTree