sujith90/Pi #18: Create an ETS reminder

BrianPearTree, 2016-01-16:
Within x minutes if time to leave, if the employee leaves desk without notifying monkey time card is filled out, give a gentle reminder!

btbam91, 2016-01-16:
[image]
[image]

Making good progress here.... Learned how to save and persist settings using .json files. Much easier than just writing to a .txt file.

btbam91, 2016-01-18:
The "Settings" aspect of this feature is implemented . The user can set the 1. Time that the user leaves work and 2. The x amount of minutes within the time time. The settings properly persist as the application is opened and closed.

btbam91, 2016-01-22:
[image]
[image]

I believe I have finished implementing this feature. I won't be able to make the meeting on Monday so I recorded a demo of the functionality in action: http://www.screencast.com/t/VygEugb7OV5

closed 2016-01-22 by btbam91

BrianPearTree, 2016-01-22:
Wow! This is really cool. Nice implementation, great demo.